udark/underdark-android #1: UDApp example doesn’t work on same devices via Bluetooth.

shahid-sabir, 2016-01-11:
HTC One M8 has used as  Android 5.x with hardware BLE Peripheral mode support. With other 2 android devices and one iPhone cross mesh network works excellent using the play store app. But  UDApp example doesn’t work on same devices via Bluetooth.

virl, 2016-01-11:
Thanks, I'll look at it.

virl, 2016-01-13:
@shahid-sabir I've re-checked on Nexus 6 (with Android 6.0.1) and Samsung Galaxy A5 (4.4) and it works fine.

shahid-sabir, 2016-01-13:
Dear Sir, I also checked and got fine. Another one thing the project is
build on API Level 22 why not it is build on API 23?

Shahid Sabir

virl, 2016-01-13:
@shahid-sabir Project is built on API 22 because in API 23 there more permission restrictions, which should be dealt separately.

Why you need to API 23?

shahid-sabir, 2016-01-13:
I faced  problem on marshmallow when i tried to use the app on it. So I thought if it is built on 23 it may not have the problems.

virl, 2016-01-13:
@shahid-sabir what problems have you faced?

shahid-sabir, 2016-01-13:
It was not getting connected with the nodes. Where other devices get connected.

virl, 2016-01-13:
@shahid-sabir Android 6 support was added only 2 days ago. Does it work now?

shahid-sabir, 2016-01-13:
I didn't check after that.

virl, 2016-01-13:
@shahid-sabir It should work now and most probably API level have nothing to do with it — it should work fine with API 22.

shahid-sabir, 2016-01-13:
So far I know Marshmallow comes with new permissions architecture. So that I thought It might create problem.

shahid-sabir, 2017-09-09:
Hi Vladimir Shavanov.
How are you? I need to talk to you regarding your underdark framework.

Regards
Sabir

virl, 2017-09-09:
@shahid-sabir Hello. I'm fine. You can send me email.

shahid-sabir, 2017-09-09:
shahid_sabir  Skype