udark/underdark-android #2: time out issue each time while sending data via WIFI

rahil1991, 2016-02-25:
java.net.ConnectException: failed to connect to /192.168.5.111 (port 50000): connect failed: ETIMEDOUT (Connection timed out)
                                                             at libcore.io.IoBridge.connect(IoBridge.java:124)
                                                             at java.net.PlainSocketImpl.connect(PlainSocketImpl.java:183)
                                                             at java.net.PlainSocketImpl.connect(PlainSocketImpl.java:163)
                                                             at java.net.Socket.startupSocket(Socket.java:590)
                                                             at java.net.Socket.tryAllAddresses(Socket.java:128)
                                                             at java.net.Socket.<init>(Socket.java:178)
                                                             at java.net.Socket.<init>(Socket.java:150)
                                                             at io.underdark.util.nslogger.NSLogger.connectImpl(SourceFile:163)
                                                             at io.underdark.util.nslogger.NSLogger.access$100(SourceFile:40)
                                                             at io.underdark.util.nslogger.NSLogger$2.run(SourceFile:152)
                                                             at android.os.Handler.handleCallback(Handler.java:739)
                                                             at android.os.Handler.dispatchMessage(Handler.java:95)
                                                             at android.os.Looper.loop(Looper.java:135)
                                                             at io.underdark.util.dispatch.DispatchQueue$2.run(SourceFile:82)
                                                             at java.lang.Thread.run(Thread.java:818)
                                                          Caused by: android.system.ErrnoException: connect failed: ETIMEDOUT (Connection timed out)
                                                             at libcore.io.Posix.connect(Native Method)
                                                             at libcore.io.BlockGuardOs.connect(BlockGuardOs.java:111)
                                                             at libcore.io.IoBridge.connectErrno(IoBridge.java:137)
                                                             at libcore.io.IoBridge.connect(IoBridge.java:122)
                                                             at java.net.PlainSocketImpl.connect(PlainSocketImpl.java:183) 
                                                             at java.net.PlainSocketImpl.connect(PlainSocketImpl.java:163) 
                                                             at java.net.Socket.startupSocket(Socket.java:590) 
                                                             at java.net.Socket.tryAllAddresses(Socket.java:128) 
                                                             at java.net.Socket.<init>(Socket.java:178) 
                                                             at java.net.Socket.<init>(Socket.java:150) 
                                                             at io.underdark.util.nslogger.NSLogger.connectImpl(SourceFile:163) 
                                                             at io.underdark.util.nslogger.NSLogger.access$100(SourceFile:40) 
                                                             at io.underdark.util.nslogger.NSLogger$2.run(SourceFile:152) 
                                                             at android.os.Handler.handleCallback(Handler.java:739) 
                                                             at android.os.Handler.dispatchMessage(Handler.java:95) 
                                                             at android.os.Looper.loop(Looper.java:135) 
                                                             at io.underdark.util.dispatch.DispatchQueue$2.run(SourceFile:82) 
                                                             at java.lang.Thread.run(Thread.java:818)

virl, 2016-02-25:
Please describe conditions under which this issue appears.

rahil1991, 2016-02-26:
This issue occurs upto 7 times out of ten when we try to send data from one device to another.After calling sendFrame() .

virl, 2016-02-26:
Please describe what devices are you using, what android versions, what wi-fi config, and in what state is the app on the devices. Does issue reproduces with demo UDApp?

Come on people, I'm not a telepath. :)