udark/underdark-android #3: Transmission between android device is not working

rahil1991, 2016-03-19:
each time i got this,

D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=5

i use moto g,Hp tablet ,Samsung s4 for testing.

virl, 2016-03-19:
@rahil1991 Android bluetooth is very buggy on particular devices.
Thanks for the device models — most probably they have device-specific Bluetooth bugs.

If you happen to fix the issue or find workaround, feel free to submit pull request.

mrp14, 2016-05-24:
Same thing here on Asus Zenfone 2 Laser (ASUS_Z00ED).
Maybe it's because we should call all BLE interaction code in UIThread, as stated here : http://stackoverflow.com/a/20507449/2806497 ?

virl, 2016-05-24:
@mrp14 Library is specifically designed to call all Bluetooth API methods only on main threads (except for reading/writing of BluetoothSocket, which is explicitly permitted to be used from another threads).

But, of course, maybe there is a bug somewhere in the code of the library.

mrp14, 2016-05-24:
I finally tried and it changes nothing.

Here is what i'm getting in the log : 

> bt accept() failed for uuid [[UUID]] java.io.IOException: read failed, socket might closed or timeout, read ret: -1

virl, 2016-05-24:
@mrp14 Due to Android models fragmentation and very unstable Bluetooth stack (prior to 7.0), it can either be a bug in stack itself or the library. Anyway, it can be debugged and fixed only using real device that it occurs on.

mrp14, 2016-05-24:
That's exactly what i'm trying to do, maybe it would be useful for others too. Do you have any clue on how I can investigate this IOException ? 
I'm gonna try some others BLE librairies and demo apps on this device to see if there is a way to get it working.

virl, 2016-05-24:
@mrp14 Maybe it's some specific bug in that version of the BT stack that limits listening to only 3 channels. Or maybe using BLE conflicts with RFCOMM listening methods of Bluetooth API.

You can try to reproduce the bug with pure Bluetooth/BLE Android API — create same amount of listening BluetoothServerSockets and try to advertise something on BLE.

mrp14, 2016-05-24:
I'm going to try it.
BTW, is the Solidarity app source code available somewhere ? I would be interested to dig in it.

virl, 2016-05-24:
@mrp14 I've sent it to your email.

virl, 2016-05-24:
@mrp14 I've wrote message to your email.

azizimusa, 2020-08-11:
send me the solidarity source code. I want it as well