udark/underdark-android #5: manifest merger error due to android:label specified in underdark manifest

dmitry-voronkevich, 2016-06-21:
Hi,

Thank you very much for your library.

I faced an issue with manifest merger. In our case we have very complicated build system with many manifests and flavours. Once I add dependency to underdark, build fails with message about duplication of android:label attribute of an application.

There is no requirement for android library to put any attribute for application tag. There is no need to add allowBackup and label in library.

Also when users add dependency to your lib, their apps automatically become backup supportive - which might be against their needs.

Is it possible to remove them? (I can create a pull request if you wish)

virl, 2016-06-21:
@dmitry-voronkevich Hello! Thanks for your kind words.

Yes, please create pull request. I will test it and accept it if everything is fine.

Most probably these manifest attributes were added due to bug in old version of Gradle / Android build tools which I used when developing a library.

dmitry-voronkevich, 2016-06-21:
I created pull request https://github.com/udark/underdark-android/pull/6
This issue can be closed, I suppose.

Thank you for reply

closed 2016-06-21 by dmitry-voronkevich