udark/underdark-android #9: Documentation

s1mar, 2017-01-07:


SergeyBurlaka, 2017-11-07:
no documentation, yet

virl, 2017-11-07:
@SergeyBurlaka All the existing docs are in JavaDoc/ObjcDoc format inside the framework packages.
I'm currently working on major API upgrade (with fully compatibility with existing API) and planning to publish API docs as separate site for convenience.

SergeyBurlaka, 2017-11-13:
ok

sahilpatel15, 2017-11-22:
I have tried many different approaches to get JavaDocs into the project but nothing seems to work. The API is a little difficult to follow without the documentation. Can you provide any alternative way to get the docs. Also, when will this major upgrade to API roll out?

bqkhiem96, 2017-12-29:
I still don't have any idea on how to access the documentation, could you instruct us prior to this instance in a different way? @virl

Jrage, 2018-06-14:
Any progress on getting the documentation?

virl, 2018-06-15:
@Jrage Hello! I'm currently working on the beta of next version of the library. As part of improving library's API I also improving its documentation.
Intermediary docs for upcoming API are available there: http://docs.underdark.io/cocoa/latest
Old API will be fully supported.

eazuka, 2018-07-04:
@virl any update on proper documentation

virl, 2018-07-04:
@eazuka I'm currently working on the beta of next version of the library, so all documentation there is available through that link. It includes iOS docs for current version's API too.

Docs for current version are also in ObjC-doc and javadoc form inside corresponding binary frameworks and available through autocompletion in Xcode and Android Studio.