udark/underdark-android #10: shouldPairDevice always returns true?

mchlnix, 2017-01-10:
https://github.com/udark/underdark-android/blob/749278420fbdfb978adcfc0c0aca729ea623cf4a/udark/underdark/src/main/java/impl/underdark/transport/bluetooth/BtTransport.java#L296

Just saw that in the code. There seems to be something wrong here, doesn't it?

virl, 2017-01-10:
@mchlnix Thanks for the bug report. Pairing currently is not used, so this code doesn't matter currently.

mchlnix, 2017-01-10:
I'm browsing your code in the hopes of figuring out how you handle the multiple UUIDs on multiple phones.

If you have the time, could you answer me the following question?

As far as I know to establish a connection between to phones both need to use the same UUID when connecting/accepting a socket. So these must be known/constant on both phones.
The problem that seems to arise is, that one phone might have used the first 6 of 7 UUIDs to connect to phones and another used the last 6 of 7. Meaning, that, while they have a UUID free to connect, they can't because it is not the same.

What is your solution for this problem?

virl, 2017-01-10:
@mchlnix My solution to the problem is to try to connect by all of the UUIDs in random order until free one is found.

mchlnix, 2017-01-10:
Sounds like having all 7 uuids connected is very unlikely, then, or am I missing something?

virl, 2017-01-10:
@mchlnix No, I just described currently implemented solution.