udark/underdark-android #12: Problem of file transfers

Lincarl7236, 2017-02-18:
Hello, I am interested in the project named underdark-android . And I have some questions, eg"1. Could I achieve transmission of files by wifi ?On my phone , I can only transfer photo and message by bluetooth.  2.The app I used is downloaded from Goole Play.  When I used APK built by the repo downloaded from   "https://github.com/udark/underdark-android", I could not transfer anything. Would you mind giving the source code of the APP on Goole Play.

virl, 2017-02-18:
@Lincarl7236 Hello. Sorry, Solidarity app is closed source currently.
You can transfer file by sending them it in single frame (if file is small) or by splitting it into multiple frames.

Lincarl7236, 2017-03-09:
My graduation project need to design a Bluetooth ad-hoc web-chat software, I would like to know how  Solidarity app realize auto-discover and connect to each other via the Bluetooth , in other words, how do you build the ad-hoc network via the Bluetooth? Could you please give me some advice?

virl, 2017-03-09:
@Lincarl7236 Hello. It doesn't build ad-hoc network. Instead, it uses store-and-send mechanism for messages and synchronisation of them upon connect.

Lincarl7236, 2017-03-10:
@virl How Solidarity app realize auto-discover and connect to each other via the Bluetooth? I want  to achieve it, but I have no idea about it. Could you please give me some advice?

virl, 2017-03-10:
@Lincarl7236 It uses this Underdark library for this, which in turn uses iOS and Android system APIs.

Lincarl7236, 2017-03-10:
@virl If I want design a  web-chat software realized auto-discover and connect to each other via the Bluetooth. How should I apply this Underdark？

virl, 2017-03-10:
@Lincarl7236 It's up to you. Underdark allows you to transfer arbitrary data between devices.

Lincarl7236, 2017-03-27:
Hello. Sorry to bother you again, I have some questions about the code downloaded from "https://github.com/udark/underdark-android". This code contains two parts, “udapp” and“underdark” . Is udapp an example that used Uderdark library?

virl, 2017-03-27:
@Lincarl7236 Yes.

Lincarl7236, 2017-03-28:
@virl Could I achieve transmission of files using the  Underdark library ? It means that I design an android app using the Uderdark library. And  what should I do to achieve transmission of files ?

virl, 2017-03-28:
@Lincarl7236 If file is small and you don't need to track transfer progress, you can send it in single frame/packet.
Otherwise you need to split it manually to several chunks and send them consecutively.

Lincarl7236, 2017-04-10:
@virl Hello. What should I do to achieve sending files in single frame/packet.

virl, 2017-04-10:
@Lincarl7236 Just send their data via sendFrame()