udark/underdark-android #14: Transfer audio / video or any buffer

AeonDave, 2017-07-28:
Hello!
is possible to send any video or audio buffer from 1 to 1 link?

thank you

virl, 2017-07-28:
@AeonDave Hello! Yes, you can send it as bytes.

AeonDave, 2017-07-28:
Ty! but how i can manage to send a buffer?
is there any example?

virl, 2017-07-28:
@AeonDave you can to it via sendFrame() method of Link class.

AeonDave, 2017-07-29:
That's exactly what i do with common data, but a buffer is a little different, so which is the best method to send a registering audio (Buffer / byte[]) to another link with sendFrame()?

ty for your help!

D

virl, 2017-07-29:
@AeonDave just use first bytes of frame to distinguish between common data and buffer.

pcace, 2019-05-07:
Hi, have you had any success sending/receiving audio?