udark/underdark-android #15: Enable logging

hojberg, 2017-08-04:
I'm having trouble seeing any logging.
What is supposed to happen when `Underdark.configureLogging(true);` is called?

virl, 2017-08-05:
@hojberg It logs using SLF4J.

hojberg, 2017-08-07:
@virl I'm not too familiar with SLF4J. From the docs https://www.slf4j.org/android/ it seems that all I should be doing was adding `compile 'eu.lp0.slf4j:slf4j-android:1.7.25-0'` to my Gradle file.

Do you know of any additional configuration needed for Underdark?

virl, 2017-08-07:
@hojberg As I understand, think dependency just routes logging to logcat. If that's what you need, you can use it.

hojberg, 2017-08-07:
@virl I do need it in logcat, but even after adding `compile 'eu.lp0.slf4j:slf4j-android:1.7.25-0'` and `Underdark.configureLogging(true)` I still so no logs from Underdark :(

hojberg, 2017-08-08:
Got this working by configuring SLF4J correctly for DEBUG.

closed 2017-08-08 by hojberg