udark/underdark-android #16: Crash relative to Bluetooth usage

theFasta, 2017-09-11:
Hi, I am testing the library, and i noticed that it crashes when initialize it for both WIFI and Bluetooth, with ` EnumSet.of(TransportKind.BLUETOOTH, TransportKind.WIFI)`. if i start the app with BT turned off, and i switch it on, the app crashes with the following stacktrace: 

```
09-11 16:45:45.982 23118-23118/com.oimmei.cellvoice.guide E/AndroidRuntime: FATAL EXCEPTION: main
                                                                            Process: com.oimmei.cellvoice.guide, PID: 23118
                                                                            java.lang.RuntimeException: Error receiving broadcast Intent { act=android.bluetooth.adapter.action.SCAN_MODE_CHANGED flg=0x4000010 (has extras) } in impl.underdark.transport.bluetooth.BtTransport$6@8c4d11
                                                                                at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$-android_app_LoadedApk$ReceiverDispatcher$Args_51267(LoadedApk.java:1305)
                                                                                at android.app.-$Lambda$FilBqgnXJrN9Mgyks1XHeAxzSTk.$m$0(Unknown Source:4)
                                                                                at android.app.-$Lambda$FilBqgnXJrN9Mgyks1XHeAxzSTk.run(Unknown Source:0)
                                                                                at android.os.Handler.handleCallback(Handler.java:789)
                                                                                at android.os.Handler.dispatchMessage(Handler.java:98)
                                                                                at android.os.Looper.loop(Looper.java:164)
                                                                                at android.app.ActivityThread.main(ActivityThread.java:6541)
                                                                                at java.lang.reflect.Method.invoke(Native Method)
                                                                                at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:240)
                                                                                at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:767)
                                                                             Caused by: java.lang.NullPointerException: Attempt to get length of null array
                                                                                at com.google.protobuf.ByteString.copyFrom(ByteString.java:218)
                                                                                at impl.underdark.transport.bluetooth.switcher.BtSwitcherDumb.setMyAddress(BtSwitcherDumb.java:47)
                                                                                at impl.underdark.transport.bluetooth.BtTransport.onReceive_ACTION_SCAN_MODE_CHANGED(BtTransport.java:374)
                                                                                at impl.underdark.transport.bluetooth.BtTransport.onReceive(BtTransport.java:326)
                                                                                at impl.underdark.transport.bluetooth.BtTransport.access$200(BtTransport.java:49)
                                                                                at impl.underdark.transport.bluetooth.BtTransport$6.onReceive(BtTransport.java:247)
                                                                                at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$-android_app_LoadedApk$ReceiverDispatcher$Args_51267(LoadedApk.java:1295)
                                                                                at android.app.-$Lambda$FilBqgnXJrN9Mgyks1XHeAxzSTk.$m$0(Unknown Source:4) 
                                                                                at android.app.-$Lambda$FilBqgnXJrN9Mgyks1XHeAxzSTk.run(Unknown Source:0) 
                                                                                at android.os.Handler.handleCallback(Handler.java:789) 
                                                                                at android.os.Handler.dispatchMessage(Handler.java:98) 
                                                                                at android.os.Looper.loop(Looper.java:164) 
                                                                                at android.app.ActivityThread.main(ActivityThread.java:6541)

```

Your help would be really appreciated, thank you

theFasta, 2017-09-11:
Second issue, still relative tu Bluetooth. ..
If i start the App with Bluetooth already enabled with 
I obtain the following stacktrace: `EnumSet.of(TransportKind.BLUETOOTH, TransportKind.WIFI)`
```
09-11 16:47:16.234 24623-24623/com.oimmei.cellvoice.guide E/AndroidRuntime: FATAL EXCEPTION: main
                                                                            Process: com.oimmei.cellvoice.guide, PID: 24623
                                                                            java.lang.NullPointerException: Attempt to get length of null array
                                                                                at com.google.protobuf.ByteString.copyFrom(ByteString.java:218)
                                                                                at impl.underdark.transport.bluetooth.switcher.BtSwitcherDumb.setMyAddress(BtSwitcherDumb.java:47)
                                                                                at impl.underdark.transport.bluetooth.BtTransport.checkScanModeAtStart(BtTransport.java:400)
                                                                                at impl.underdark.transport.bluetooth.BtTransport.startInternal(BtTransport.java:256)
                                                                                at impl.underdark.transport.bluetooth.BtTransport$4.run(BtTransport.java:200)
                                                                                at android.os.Handler.handleCallback(Handler.java:789)
                                                                                at android.os.Handler.dispatchMessage(Handler.java:98)
                                                                                at android.os.Looper.loop(Looper.java:164)
                                                                                at android.app.ActivityThread.main(ActivityThread.java:6541)
                                                                                at java.lang.reflect.Method.invoke(Native Method)
                                                                                at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:240)
                                                                                at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:767)
```

virl, 2017-09-11:
@theFasta Hello. Thanks for the bug report.
On what Android version and device do you experience it?

theFasta, 2017-09-11:
Basically I'm working on all platforms from 19 and earlier. Today I've been working on Lollipop, Marshmallow, Nougat and Oreo...

virl, 2017-09-11:
@theFasta Okay, can you then send me minimal Android project that reproduces the issue?

theFasta, 2017-09-11:
Actually not. It's not that I don't want to share the code, but the problem verifies just when the node gets started, so it's kinda "out of my jurisdiction"...

virl, 2017-09-11:
@theFasta You don't have to share code of your proprietary app, you can just make separate minimal project that demonstrates just the issue.
I have other people that use library and they didn't report this problem. So maybe there is something specific in how you start it or in code context in which you start it.

theFasta, 2017-09-11:
```
package com.oimmei.cellvoice.guide.model

import android.content.Context
import com.oimmei.cellvoice.guide.utils.Constants
import io.underdark.Underdark
import io.underdark.transport.Link
import io.underdark.transport.Transport
import io.underdark.transport.TransportKind
import io.underdark.transport.TransportListener
import java.util.*

/**
 * Created by Andrea Fastame
 *
 * @email andrea@oimmei.com
 * @since 06/05/17 - 12:38
 * Copyright Oimmei Srls 2015,2016 - www.oimmei.com
 */

class Node(context: Context, private val listener: NodeListener) : TransportListener {
    private var running: Boolean = false
    private var nodeId: Long = 0
    private val transport: Transport

    enum class FRAME_TYPE private constructor(val frameType: Int, val stringVal: String) {
        EOF(0, "0"), AUDIO(1, "1"), TEXT(2, "2");

        val byteVal: Byte

        init {
            byteVal = java.lang.Byte.decode(stringVal)!!

        }

    }

    val links = ArrayList<Link>()
    var framesCount = 0
        private set

    init {

//        do {
        nodeId = Random().nextLong()
//        } while (nodeId == 0L)

        if (nodeId < 0)
            nodeId = -nodeId

        configureLogging()

//        val kinds = EnumSet.of(TransportKind.BLUETOOTH, TransportKind.WIFI)
        val kinds = EnumSet.of(TransportKind.WIFI)
        //kinds = EnumSet.of(TransportKind.BLUETOOTH);

        this.transport = Underdark.configureTransport(
                Constants.NODE_APP_ID,
                nodeId,
                this,
                null,
                context,
                kinds
        )
    }

    private fun configureLogging() {
        //        NSLoggerAdapter adapter = (NSLoggerAdapter)
        //                StaticLoggerBinder.getSingleton().getLoggerFactory().getLogger(Node.class.getName());
        //        adapter.logger = new NSLogger(listener.getApplicationContext());
        //        adapter.logger.connect("192.168.5.203", 50000);

        Underdark.configureLogging(true)
    }

    fun start() {
        if (running)
            return

        running = true
        transport.start()
    }

    fun stop() {
        if (!running)
            return

        running = false
        transport.stop()
    }

    fun broadcastFrame(frameData: ByteArray) {
        if (links.isEmpty())
            return

        ++framesCount
        listener.refreshPeers(links.size)

        for (link in links)
            link.sendFrame(frameData)
    }

    //region TransportListener
    override fun transportNeedsActivity(transport: Transport, callback: TransportListener.ActivityCallback) {
        //        callback.accept(listener);
    }

    override fun transportLinkConnected(transport: Transport, link: Link) {
        links.add(link)
        listener.refreshPeers(links.size)
    }

    override fun transportLinkDisconnected(transport: Transport, link: Link) {
        links.remove(link)
        listener.refreshPeers(links.size)

        if (links.isEmpty()) {
            framesCount = 0
            listener.refreshFrames(framesCount)
        }
    }

    override fun transportLinkDidReceiveFrame(transport: Transport, link: Link, frameData: ByteArray) {
        ++framesCount
        listener.refreshFrames(framesCount)
        listener.newFrame(frameData)
    }

    interface NodeListener {
        fun refreshFrames(frameCount: Int)

        fun refreshPeers(linksSize: Int)

        fun newFrame(message: ByteArray)
    }
    //endregion {
}
```

virl, 2017-09-11:
@theFasta Thanks, I will look into it.

virl, 2017-11-17:
I've released Underdark Android 1.0.12 with crash fix for Android 8.

theFasta, 2017-11-22:
Hi!

Thanks alot, I will try the new version and let you know.

Cheers

Fasta thinks different
http://www.yourprofolio.com/thefasta
http://fastaproductions.imagekind.com