udark/underdark-android #20: Trying to unconnect

SergeyBurlaka, 2017-11-13:
First i create conection:
my logs

```
11-13 18:51:40.776 6259-6259/com.jellyworkz.udark D/Node: transportLinkConnected Link = alink(1) nodeId 7855185628251833254
11-13 18:51:40.776 6259-6259/com.jellyworkz.udark D/Node: ----------------------$$--------------$$--------
11-13 18:51:40.876 6259-6259/com.jellyworkz.udark D/Node: transportLinkConnected Link = alink(1) nodeId 4618952656116501996
11-13 18:51:40.876 6259-6259/com.jellyworkz.udark D/Node: ----------------------$$--------------$$--------

```
I try unconnect via this method 

```
private val mLinks = ArrayList<Link>()`
fun unconnected() {
        for (link in mLinks) {
            link.disconnect()
        }
        dataTransfer = null
    }

```
disconnect() is method from Underdark:

```
public void disconnect() {
        this.transport.queue.dispatch(new Runnable() {
            public void run() {
                Iterator var1 = AggLink.this.links.iterator();

                while(var1.hasNext()) {
                    Link link = (Link)var1.next();
                    link.disconnect();
                }

            }
        });
    }
```
My logs:

```
11-13 18:53:20.963 6259-6259/com.jellyworkz.udark D/Node: transportLinkDisconnected  Link = alink(0) nodeId 7855185628251833254
11-13 18:53:20.963 6259-6259/com.jellyworkz.udark D/Node: ----------------------$$--------------$$--------
11-13 18:53:20.963 6259-6259/com.jellyworkz.udark D/Node: transportLinkDisconnected  Link = alink(0) nodeId 4618952656116501996
11-13 18:53:20.963 6259-6259/com.jellyworkz.udark D/Node: ----------------------$$--------------$$--------
```

Then i conect again by create new Node 
` mNodeSound = Node(this, this, Node.NodeType.SOUND, SOUND_ID_PREFIX + id)`

It works
But create  redundant links:
See my logs:

```
11-13 18:54:14.946 6259-6259/com.jellyworkz.udark D/Node: transportLinkConnected Link = alink(1) nodeId 7376268880458696257
11-13 18:54:14.946 6259-6259/com.jellyworkz.udark D/Node: ----------------------$$--------------$$--------
11-13 18:54:14.966 6259-6259/com.jellyworkz.udark D/Node: transportLinkConnected Link = alink(0) nodeId 755959319961052674
11-13 18:54:14.966 6259-6259/com.jellyworkz.udark D/Node: ----------------------$$--------------$$--------
11-13 18:54:14.966 6259-6259/com.jellyworkz.udark D/Node: transportLinkDisconnected  Link = alink(0) nodeId 755959319961052674
11-13 18:54:14.966 6259-6259/com.jellyworkz.udark D/Node: ----------------------$$--------------$$--------
11-13 18:54:14.976 6259-6259/com.jellyworkz.udark D/Node: transportLinkDisconnected  Link = alink(0) nodeId 7376268880458696257
11-13 18:54:14.976 6259-6259/com.jellyworkz.udark D/Node: ----------------------$$--------------$$--------
11-13 18:54:15.066 6259-6259/com.jellyworkz.udark D/Node: transportLinkConnected Link = alink(1) nodeId 3420549108642149074
11-13 18:54:15.066 6259-6259/com.jellyworkz.udark D/Node: ----------------------$$--------------$$--------
11-13 18:54:15.096 6259-6259/com.jellyworkz.udark D/Node: transportLinkConnected Link = alink(0) nodeId 5710137947420927346
11-13 18:54:15.106 6259-6259/com.jellyworkz.udark D/Node: ----------------------$$--------------$$--------
11-13 18:54:15.106 6259-6259/com.jellyworkz.udark D/Node: transportLinkDisconnected  Link = alink(0) nodeId 5710137947420927346
11-13 18:54:15.106 6259-6259/com.jellyworkz.udark D/Node: ----------------------$$--------------$$--------
11-13 18:54:15.116 6259-6259/com.jellyworkz.udark D/Node: transportLinkDisconnected  Link = alink(0) nodeId 3420549108642149074
11-13 18:54:15.116 6259-6259/com.jellyworkz.udark D/Node: ----------------------$$--------------$$--------
11-13 18:54:15.677 6259-6259/com.jellyworkz.udark D/Node: transportLinkConnected Link = alink(1) nodeId 7855185628251833254
11-13 18:54:15.677 6259-6259/com.jellyworkz.udark D/Node: ----------------------$$--------------$$--------

```

How top fix it?

virl, 2017-11-13:
@SergeyBurlaka Hello. Can you describe what's the problem in more detail?
What do you mean by "redundant links"?

SergeyBurlaka, 2017-11-14:
I create connection betwen only 2 device via 2 nodes So i need only 2 links connected
So i watch how many links connect by logs here:
```
 //on connection
    override fun transportLinkConnected(transport: Transport, link: Link) {
        mLinks.add(link)

        Log.d(TAG, "transportLinkConnected Link = $link")
      
        when (type) {
            NodeType.MESSAGE -> dataTransfer?.onRefreshedPeersMessage(mLinks.size)

            NodeType.SOUND -> dataTransfer?.onRefreshedPeersSound(mLinks.size)
        }
    }
```

For connection i created Node object, here my Kotlin code:

```
class Node(context: Context, private var dataTransfer: DataTransfer?, private val type: NodeType, appId: Int)
    : TransportListener

 init {
        do {
            mNodeId = Random().nextLong()
        } while (mNodeId == 0L)

        if (mNodeId < 0)
            mNodeId = -mNodeId

        configureLogging()
        val kinds = EnumSet.of(TransportKind.WIFI)

        this.mTransport = Underdark.configureTransport(
                appId, mNodeId, this, null,
                context.applicationContext, kinds
        )
    }
```

And first time see 2 connected links in logs:

```
..../Node: transportLinkConnected Link = alink(1) nodeId 7855185628251833254
--
... D/Node: transportLinkConnected Link = alink(1) nodeId 4618952656116501996
```

It is work greate.

But  then i try unconnect, 

```
public void disconnect() {
        this.transport.queue.dispatch(new Runnable() {
            public void run() {
                Iterator var1 = AggLink.this.links.iterator();

                while(var1.hasNext()) {
                    Link link = (Link)var1.next();
                    link.disconnect();
                }

            }
        });
    }
```
and connect again 

And quantity of connection links is growing:

```
//Connected
...D/Node: transportLinkConnected Link = alink(1) nodeId 7376268880458696257
--
..... D/Node: transportLinkConnected Link = alink(0) nodeId 755959319961052674
--
//Disconnected
...D/Node: transportLinkDisconnected  Link = alink(0) nodeId 755959319961052674
--
...D/Node: transportLinkDisconnected  Link = alink(0) nodeId 7376268880458696257

Connected
.... D/Node: transportLinkConnected Link = alink(1) nodeId 3420549108642149074

.....D/Node: transportLinkConnected Link = alink(0) nodeId 5710137947420927346

//Disconnected
.... D/Node: transportLinkDisconnected  Link = alink(0) nodeId 5710137947420927346

.....D/Node: transportLinkDisconnected  Link = alink(0) nodeId 3420549108642149074

....D/Node: transportLinkConnected Link = alink(1) nodeId 7855185628251833254
```

This is because the old links are not completely destroyed, but new links are created.

So how i must disconnect the botch devices?

SergeyBurlaka, 2017-11-14:
> You can implment disconnection business logic on top of connected links.
So, you meen it not necessary to call link.disconnect()*, instead i can use some status flag. 

*disconnect() is method from Underdark:

```
public void disconnect() {
        this.transport.queue.dispatch(new Runnable() {
            public void run() {
                Iterator var1 = AggLink.this.links.iterator();

                while(var1.hasNext()) {
                    Link link = (Link)var1.next();
                    link.disconnect();
                }

            }
        });
    }
```

virl, 2017-11-14:
> So, you meen it not necessary to call link.disconnect()*, instead i can use some status flag.

Yes, you can implement your logical disconnection on top of long-lived links.

SergeyBurlaka, 2017-11-14:
I think now everything became very clear. Thanks

closed 2017-11-14 by SergeyBurlaka