udark/underdark-android #21: Bluetooth LE advertising doesn't work even with Peripheral mode Device

Buddabread, 2017-12-07:
Hello,

this Framework works great with Wifi, but it's not possible to connect devices over Bluetooth.
The devices can't find each other, do you have any idea for a solution?

Cheers
Budda

virl, 2017-12-07:
@Buddabread What Android version do you use?

Buddabread, 2017-12-08:
@virl Android 6.0 without LE Periperal and Android 8.1 which supports Periperal Mode

virl, 2017-12-08:
@Buddabread Android 8 currently can only connect to pre-Android8 devices.
But for that pre-Android8 device should support BLE Peripheral mode.

Buddabread, 2018-01-04:
With Android Version <8 Bluetooth LE works great. 
Do you have any updates scheduled to support Android 8?

virl, 2018-01-05:
@Buddabread It will be known in nearby months, depending on quality of BLE stack in Android 8.x

Buddabread, 2018-04-24:
Are there any news with the quality of the BLE stack in Android 8?

virl, 2018-04-25:
@Buddabread Yes, BLE improves in Android 8, but via Bluetooth Android 8 devices can only connect to pre-Android 8 devices for now.

normdoow, 2018-06-20:
Thanks for the great library! When will support for Android 8 devices be added?

Riaxter, 2018-07-01:
Your library is super cool, but still there are some connection problem. 
HUAWEI H Phone(CAM-L32 API 23) and Samsung Galaxy S7(SM-G930K API 26) cannot discover each other, Is this same issue??

virl, 2018-07-01:
@Riaxter Thanks.
Does this phones have BLE Peripheral hardware support?
It can be checked via https://play.google.com/store/apps/details?id=com.kyriakosalexandrou.bluetoothsupportcheck&hl=en

virl, 2018-08-21:
@normdoow Greetings!
Currently I'm working on improving iOS version.
As for Android, Google broke/deprecated the Bluetooth API that I used, so it will require some time to find and test alternatives.

normdoow, 2018-08-21:
Thanks @virl!

normdoow, 2018-12-17:
@virl What is the status of this update for Android?