udark/underdark-android #23: bluetooth address null on Android OREO

skpaik, 2018-05-07:
Hi,
On Android, Oreo Underdark library cannot get the **bluetooth_address** with the folllowing methods 
`  String macAddress = android.provider.Settings.Secure.getString(context.getContentResolver(), "bluetooth_address");
` 

Please give us a update.

Thank you

skpaik, 2018-05-08:
Thank you @virl .
Please give a comment here when updates come.

virl, 2018-05-08:
@skpaik This update is already published and available in binary form in bintray repository.

skpaik, 2018-05-09:
Hi @virl Thank you.
1. **With Bluetooth** when I test on my MotoZPlay (Android OREO) of basic project you provide, 
- [ ] - I cannot discover anyone.
- [ ] - Other devices (Android Marshmellow or lower) also cannot discover MotoZPlay.
- [ ] - But all other devices are discovered each other.

2. **With Wifi** hotspot device does not discover to anyone. Suppose

- [ ] - Device A (Hotspot)
-  [ ] - Device B (Wifi on and connected to **Device A**)
-  [ ] - Device C (Wifi on and connected to **Device A**)
 Now **Device B** & **Device C** Discover to each other, but **Device A** does not discover no any one.

Please give us the updates.

Thanks in Advanced.

virl, 2018-05-10:
@skpaik In (1) case it seems some problem of MotoZPlay Bluetooth hardware/drivers: https://forums.lenovo.com/t5/Moto-Z-Play/Moto-Z-Play-Bluetooth-Connectivity-Issues/td-p/3565909

In (2) case it is limitation of Android hotspot mode in general.

tamirkonor, 2019-01-20:
this version still not supporting the library on oreo on samsung as well